agilexrobotics/piper_ros #2: URDF visual meshes incorrect?

trzy, 2024-12-01:
I've tried loading the URDF with both pybullet and klampt but the result always seems disjointed and incorrect. I'm not sure what the issue is but I suspect it is with the URDF file. I loaded the link STL and DAE files in Blender and the origin points seemed reasonable. Here is how it looks in both of the aforementioned tools.
[image]

I think the kinematic chain itself is ok because plotting the joints and links in roboticstoolbox and my own viewer appears fine.

bmagyar, 2024-12-07:
Which branch? I've tested things from the Humble branch today, running on Jazzy (hah!) and it was fine.

[image]

RosenYin, 2025-01-03:
我修改了新的urdf文件，并修改了末端的坐标系方向，令其与机械臂底层sdk反馈的endpose对齐；还附着了颜色，可以查看最新的urdf

trzy, 2025-01-15:
Confirmed working with new piper_description.urdf!
[image]

closed 2025-01-15 by trzy