agilexrobotics/piper_ros #7: TypeError: __new__() got an unexpected keyword argument 'can_name'

Akumar201, 2025-01-29:
Hello , I am trying to follow your documentation. Here is the error when I am trying to run 

`
rosrun piper piper_ctrl_single_node.py _can_port:=can0 _mode:=0`

```
[INFO] [1738190961.994531]: /piper_ctrl_single_node_1551_1738190961871/auto_enable is False
[INFO] [1738190961.996305]: /piper_ctrl_single_node_1551_1738190961871/girpper_exist is True
[INFO] [1738190961.998096]: /piper_ctrl_single_node_1551_1738190961871/rviz_ctrl_flag is False
Traceback (most recent call last):
  File "/root/maxinsights_ws/src/Piper_ros/src/piper/scripts/piper_ctrl_single_node.py", line 434, in <module>
    piper_signle = C_PiperRosNode()
  File "/root/maxinsights_ws/src/Piper_ros/src/piper/scripts/piper_ctrl_single_node.py", line 80, in __init__
    self.piper = C_PiperInterface(can_name=self.can_port)
TypeError: __new__() got an unexpected keyword argument 'can_name'

```

This is my sdk version 
`piper-sdk                 0.1.9  `

wujiashu0421, 2025-01-29:
Hi, I had the exact same problem yesterday.

I was able to bypass the issue by modifying `Piper_ros/src/piper/scripts/piper_ctrl_single_node.py` and deleting the `can-name=self.can_port` in the piper class instantiation. You would need to modify the CAN number if it's not can0 which is the default.

I suspect the class definition has been modified in the SDK and the instantiation has not changed. @RosenYin 

George

Akumar201, 2025-01-29:
@wujiashu0421 Thank you for your response. I am summarizing it, anyone having this issue you need to change `self.piper = C_PiperInterface(can_name=self.can_port)` to `self.piper = C_PiperInterface()`

wujiashu0421, 2025-01-29:
Hi, I would still prefer to leave the issue open since it is not the expected behaviour between the ROS package and SDK.

RosenYin, 2025-01-30:
这个是因为interface中一个__new__函数的形参与init函数形参不同导致的，尝试进入到piper_sdk路径，然后进入piper_interface.py文件，将__new__函数的第一个形参改为can_name

RosenYin, 2025-01-30:
你可以尝试一下，如果有效果，请给予反馈，注意：ros的那一部分创建interface实例的传入参数要保留。

Akumar201, 2025-01-30:
@RosenYin thank you for your response, it would be great if we can keep the language of conversion to English as many people might need to read it. Thank You

wujiashu0421, 2025-01-30:
**Translation**

Rosen suggested changing the can-bus related variable name in the instantiator of the python SDK for `C_PiperInterface` from `can_port` to `can_name`, where **keeping the can variable input in ROS/ROS2 interface**.

**Solution**

The naming seems to be the issue. Created pull request in Piper_SDK (Can name fix #13)

Before the pull request is approved, one can change the name manually by the following steps

- Locate the path of the piper SDK package by `pip show piper_sdk` and cd to the library directory
- `cd piper_sdk/interface`
- Change `can_port` to `can_name` in line 337 and 345 respectively in `piper_interface.py`
- `catkin build` or `colcon build` again in ROS/ROS2 workspace.

Akumar201, 2025-02-05:
As they have merged your pull request things , should be fine by just using pip install piper_sdk==0.1.9. Thank You @wujiashu0421

closed 2025-02-05 by Akumar201