agilexrobotics/piper_ros #8: How to change the arm to follower mode

Akumar201, 2025-01-29:
Hello , I am following the Readme to run the robotics arm, I want to send commands to the robotic arm and control it, however right now I have two issues , 
1. When I am running 
```
rosservice call /enable_srv "enable_request: true"
ERROR: Unable to load type [piper_msgs/Enable].
Have you typed 'make' in [piper_msgs]?

```
I am getting the above error 
2. When running the next commands

```
rostopic pub /joint_states sensor_msgs/JointState "header:
  seq: 0
  stamp: {secs: 0, nsecs: 0}
  frame_id: ''
name: ['']
position: [0.2,0.2,-0.2,0.3,-0.2,0.5,0.01]
velocity: [0,0,0,0,0,0,10]
effort: [0,0,0,0,0,0,0.5]" 
```

The arm is not responding , I feel like this is something to do with mode in which the arm is by default. Any help would be much appriciated

wujiashu0421, 2025-01-29:
I have not tried 1. My command sequence for the arm is as follows:

1. Initialise the ros2 controller node by `roslaunch piper start_single_piper.launch` without RVIZ joint state publisher. It defaults to the robotic arm being enabled, so you can double-check the enable status by trying to move the joints. If they are locked then they are enabled.
2. I ran my custom joint state publisher, which publishes the joint angle information in /joint_states topic. You could use `rostopic pub /joint_states sensor_msgs/JointState "header:
  seq: 0
  stamp: {secs: 0, nsecs: 0}
  frame_id: ''
name: ['']
position: [0.2,0.2,-0.2,0.3,-0.2,0.5,0.01]
velocity: [0,0,0,0,0,0,10]
effort: [0,0,0,0,0,0,0.5]"`

Akumar201, 2025-02-03:
@wujiashu0421 , do you have any suggestion how to connect the arm to personal computer , the issue I am facing is the CAN module is connected to my computer but it is not showing up, I am incline toward an idea that there is a udev rules missing , but it's not in the documentation. 

When I do `lsusb` I can see  
```
Bus 001 Device 011: ID 1d50:606f OpenMoko, Inc. Geschwister Schneider CAN adapter
Bus 001 Device 012: ID 1d50:606f OpenMoko, Inc. Geschwister Schneider CAN adapter

```
`
ip link show type can` is not showing anything . 

Any help would be much appreciated

Akumar201, 2025-02-03:
Hello , so about my setup, I have connected left slave and leader arm together to a can module  , similarly I have attached right slave and leader arm together. I am following this [document](https://drive.google.com/file/d/1ab8y21zmv7GAnhaGPnWwYm3oHbWVhcgO/view?usp=sharing), when I run can_config.h  I am getting 
`Error: The number of detected CAN modules (0) does not match the expected number (3).`, even my computer is not able to see it when I do ifconfig, I feel like it is reading usb as lsusb shows 

```
Bus 001 Device 011: ID 1d50:606f OpenMoko, Inc. Geschwister Schneider CAN adapter
Bus 001 Device 012: ID 1d50:606f OpenMoko, Inc. Geschwister Schneider CAN adapter
```
but not linking to can. I am using ubuntu 22.04

wujiashu0421, 2025-02-03:
Hi, I think there's detailed instruction in this repo's readme. I'm on the
ros2 humble branch. The process involves calling can_config bash script in
different ways.

wujiashu0421, 2025-02-03:
As per the readme, you should:

1. plug in one arm only
2. Discover usb bus address by `sudo ethtool -i can0 | grep bus`
3. Activate can bus by `bash can_activate.sh can_piper 1000000 "1-2:1.0"`

Note it does not necessarily needs to be can0. The second arm you plugged in should be can1.

Akumar201, 2025-02-03:
This documenation revolves around there industrial computer in which many things are already set, when I connect my can module and run it , it's not even coming up 
```

sudo ethtool -i can0 | grep bus
Cannot get driver information: No such device

```

even when I do 

` ls -l /dev/can*` to find the name of can devices it's not coming up. Probably my computer is not detective the usb hub as can device automatically.

Akumar201, 2025-02-03:
@wujiashu0421 , there was issue with my USB hub, when I directly place it in my computer it's showing up. Now when I am trying to run the  `roslaunch piper start_ms_piper.launch mode:=0 auto_enable:=false`

I am not sure why 
```

File exists: '/root/.ros/log/5e0024e8-e289-11ef-b416-e0d46488af9a' -> '/root/.ros/log/latest'
[INFO] [1738626509.492637]: /piper_right_max_ROG_Strix_G533QR_463_7177781491788723743/can_port is can_right
[INFO] [1738626509.492739]: /piper_left_max_ROG_Strix_G533QR_463_3804503202252381110/can_port is can_left
[INFO] [1738626509.494109]: /piper_left_max_ROG_Strix_G533QR_463_3804503202252381110/mode is 0
[INFO] [1738626509.494387]: /piper_right_max_ROG_Strix_G533QR_463_7177781491788723743/mode is 0
[INFO] [1738626509.495320]: /piper_left_max_ROG_Strix_G533QR_463_3804503202252381110/auto_enable is False
[INFO] [1738626509.495821]: /piper_right_max_ROG_Strix_G533QR_463_7177781491788723743/auto_enable is False
Traceback (most recent call last):
  File "/app/cobot_magic/Piper_ros_private-ros-noetic/src/piper/scripts/piper_start_ms_node.py", line 336, in <module>
    piper_ms = C_PiperRosNode()
  File "/app/cobot_magic/Piper_ros_private-ros-noetic/src/piper/scripts/piper_start_ms_node.py", line 82, in __init__
    self.piper = C_PiperInterface()
  File "/usr/local/lib/python3.8/dist-packages/piper_sdk/interface/piper_interface.py", line 365, in __init__
    self.__arm_can=C_STD_CAN(can_name, "socketcan", 1000000, judge_flag, can_auto_init, self.ParseCANFrame)
  File "/usr/local/lib/python3.8/dist-packages/piper_sdk/hardware_port/can_encapsulation.py", line 59, in __init__
    self.JudgeCanInfo()
  File "/usr/local/lib/python3.8/dist-packages/piper_sdk/hardware_port/can_encapsulation.py", line 117, in JudgeCanInfo
Traceback (most recent call last):
  File "/app/cobot_magic/Piper_ros_private-ros-noetic/src/piper/scripts/piper_start_ms_node.py", line 336, in <module>
    raise ValueError(f"CAN socket {self.channel_name} does not exist.")
ValueError: CAN socket can0 does not exist.
    piper_ms = C_PiperRosNode()
  File "/app/cobot_magic/Piper_ros_private-ros-noetic/src/piper/scripts/piper_start_ms_node.py", line 82, in __init__
    self.piper = C_PiperInterface()
  File "/usr/local/lib/python3.8/dist-packages/piper_sdk/interface/piper_interface.py", line 365, in __init__
    self.__arm_can=C_STD_CAN(can_name, "socketcan", 1000000, judge_flag, can_auto_init, self.ParseCANFrame)
  File "/usr/local/lib/python3.8/dist-packages/piper_sdk/hardware_port/can_encapsulation.py", line 59, in __init__
    self.JudgeCanInfo()
  File "/usr/local/lib/python3.8/dist-packages/piper_sdk/hardware_port/can_encapsulation.py", line 117, in JudgeCanInfo
    raise ValueError(f"CAN socket {self.channel_name} does not exist.")
ValueError: CAN socket can0 does not exist.
CAN bus connection was not properly initialized.
CAN bus connection was not properly initialized.
[piper_left_max_ROG_Strix_G533QR_463_3804503202252381110-2] process has died [pid 518, exit code 1, cmd /app/cobot_magic/Piper_ros_private-ros-noetic/src/piper/scripts/piper_start_ms_node.py /puppet/joint_states:=/puppet/joint_left /master/joint_states:=/master/joint_left __name:=piper_left_max_ROG_Strix_G533QR_463_3804503202252381110 __log:=/root/.ros/log/5e0024e8-e289-11ef-b416-e0d46488af9a/piper_left_max_ROG_Strix_G533QR_463_3804503202252381110-2.log].
log file: /root/.ros/log/5e0024e8-e289-11ef-b416-e0d46488af9a/piper_left_max_ROG_Strix_G533QR_463_3804503202252381110-2*.log
[piper_right_max_ROG_Strix_G533QR_463_7177781491788723743-3] process has died [pid 519, exit code 1, cmd /app/cobot_magic/Piper_ros_private-ros-noetic/src/piper/scripts/piper_start_ms_node.py /puppet/joint_states:=/puppet/joint_right /master/joint_states:=/master/joint_right __name:=piper_right_max_ROG_Strix_G533QR_463_7177781491788723743 __log:=/root/.ros/log/5e0024e8-e289-11ef-b416-e0d46488af9a/piper_right_max_ROG_Strix_G533QR_463_7177781491788723743-3.log].
log file: /root/.ros/log/5e0024e8-e289-11ef-b416-e0d46488af9a/piper_right_max_ROG_Strix_G533QR_463_7177781491788723743-3*.log
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...

```

RosenYin, 2025-04-10:
### 2.1 寻找can模块

执行

```bash
bash find_all_can_port.sh
```

输入密码后，如果can模块插入了电脑，并被电脑检测到，输出类似如下：

```bash
Both ethtool and can-utils are installed.
Interface can0 is connected to USB port 3-1.4:1.0
```

如果有多个，输出类似如下：

```bash
Both ethtool and can-utils are installed.
Interface can0 is connected to USB port 3-1.4:1.0
Interface can1 is connected to USB port 3-1.1:1.0
```

有多少个can模块就会有多少行类似`Interface can1 is connected to USB port 3-1.1:1.0`的输出

其中`can1`是系统找到的can模块名字，`3-1.1:1.0`是该can模块所链接的usb端口

如果之前已经激活过can模块并其名为其它名字，这里假设名字为`can_piper`则输出如下

```bash
Both ethtool and can-utils are installed.
Interface can_piper is connected to USB port 3-1.4:1.0
Interface can0 is connected to USB port 3-1.1:1.0
```

如果没有检测到can模块，则只会输出如下：

```bash
Both ethtool and can-utils are installed.
```

### 2.2 激活单个can模块, **此处使用`can_activate.sh`脚本**

激活单个can模块的情况分两种，一种是电脑只链接了一个can模块，一种是多个can模块插入电脑但是只激活其中一个。

#### 2.2.1 pc只插入一个usb转can模块

直接执行

```bash
bash can_activate.sh can0 1000000
```

此处`can0`可以改为任意名字，`1000000`为波特率，不可更改

#### 2.2.2 pc插入多个usb转can模块， 但每次只激活一个can模块

注： 此处用于同时使用机械臂和底盘的的情况

(1) 查看can模块插在usb端口的硬件地址。拔掉所有can模块，只将连接到机械臂的can模块插入PC，执行

```shell
bash find_all_can_port.sh
```

并记录下`USB port`的数值，例如`3-1.4:1.0`

(2) 激活can设备。假设上面的`USB port`数值为`3-1.4:1.0`，执行：

```bash
bash can_activate.sh can_piper 1000000 "3-1.4:1.0"
```

解释：**3-1.4:1.0硬件编码的usb端口插入的can设备，名字被重命名为can_piper，设定波特率为1000000，并激活**

(3) 检查是否激活成功

执行`ifconfig`查看是否有`can_piper`，如果有则can模块设置成功

### 2.3 同时激活多个can模块，**此处使用`can_config.sh`脚本**

首先确定有多少个官方can模块被插入到电脑，这里假设是2

注：**若当前电脑插入了5个can模块，则需要将这5个模块都填入下述的文件，否则脚本会给出激活的模块数量与系统模块数量不一致的错误**

#### 2.3.1 记录每个can模块对应的usb口硬件地址

逐个拔插can模块并一一记录每个模块对应的usb口硬件地址

在`can_config.sh`中，`EXPECTED_CAN_COUNT`参数为想要激活的can模块数量，现在假设为2

(1) 然后can模块中的其中一个单独插入PC，执行

```shell
bash find_all_can_port.sh
```

并记录下`USB port`的数值，例如`3-1.4:1.0`

(2) 接着插入下一个can模块，注意**不可以**与上次can模块插入的usb口相同，然后执行

```shell
bash find_all_can_port.sh
```

注：**如果未曾激活过，则第一个插入的can模块会默认是can0，第二个为can1，若激活过，名字为之前激活过的名称**

#### 2.3.2 预定义USB 端口、目标接口名称及其比特率

假设上面的操作记录的`USB port`数值分别为`3-1.4:1.0`、`3-1.1:1.0`，则将下面的`USB_PORTS["1-9:1.0"]="can_left:1000000"`的中括号内部的双引号内部的参数换为`3-1.4:1.0`和`3-1.1:1.0`.

最终结果为：

`USB_PORTS["3-1.4:1.0"]="can_left:1000000"`

`USB_PORTS["3-1.1:1.0"]="can_right:1000000"`

解释：**3-1.4:1.0硬件编码的usb端口插入的can设备，名字被重命名为can_left，波特率为1000000，并激活**

#### 2.3.3 激活can模块

执行`bash can_config.sh`

#### 2.3.4 查看can是否设置成功

执行`ifconfig`查看是不是有`can_left`和`can_right`

closed 2025-04-17 by Akumar201