agilexrobotics/piper_ros #12: AttributeError: type object 'ArmEndPose' has no attribute 'end_pose'

Akumar201, 2025-02-05:
I am following the readme now , when I ran 

`rosrun piper piper_ctrl_single_node.py _can_port:=can0 _mode:=0`

I am getting the following error 
```
[INFO] [1738789868.330095]: /piper_ctrl_single_node_816_1738789868216/can_port is can0
[INFO] [1738789868.331023]: /piper_ctrl_single_node_816_1738789868216/auto_enable is False
[INFO] [1738789868.331883]: /piper_ctrl_single_node_816_1738789868216/gripper_exist is True
[INFO] [1738789868.332698]: /piper_ctrl_single_node_816_1738789868216/rviz_ctrl_flag is False
can0  is exist
can0  is UP
can0  bitrate is  1000000
can0 bus opened successfully.
Traceback (most recent call last):
  File "/app/cobot_magic/Piper_ros_private-ros-noetic/src/piper/scripts/piper_ctrl_single_node.py", line 316, in <module>
    piper_signle.Pubilsh()
  File "/app/cobot_magic/Piper_ros_private-ros-noetic/src/piper/scripts/piper_ctrl_single_node.py", line 135, in Pubilsh
    self.PublishArmEndPose()
  File "/app/cobot_magic/Piper_ros_private-ros-noetic/src/piper/scripts/piper_ctrl_single_node.py", line 190, in PublishArmEndPose
    endpos.position.x = self.piper.ArmEndPose.end_pose.X_axis/1000000
AttributeError: type object 'ArmEndPose' has no attribute 'end_pose'

```

HoranTian, 2025-02-07:
Updating the version of piper_sdk can solve this issue.

Akumar201, 2025-02-19:
The official update has been made in piper sdk which has resolved this issue.

closed 2025-02-19 by Akumar201