agilexrobotics/piper_ros #13: [Errno 2] No such file or directory: 'ip'

Akumar201, 2025-02-18:
I am trying to run [start_single_piper.launch.py)](https://github.com/agilexrobotics/Piper_ros/blob/ros-humble-no-aloha/src/piper/launch/start_single_piper.launch.py), and getting the following error , 

 ```
ros2 launch piper start_single_piper.launch.py 
[INFO] [launch]: All log files can be found below /root/.ros/log/2025-02-18-21-44-21-724010-max-agilex-2-5698
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [piper_single_ctrl-1]: process started with pid [5699]
[piper_single_ctrl-1] [INFO] [1739915061.991686459] [piper_ctrl_single_node]: can_port is can_right
[piper_single_ctrl-1] [INFO] [1739915061.992121921] [piper_ctrl_single_node]: auto_enable is True
[piper_single_ctrl-1] [INFO] [1739915061.992702794] [piper_ctrl_single_node]: gripper_exist is True
[piper_single_ctrl-1] [INFO] [1739915061.993116885] [piper_ctrl_single_node]: rviz_ctrl_flag is False
[piper_single_ctrl-1] can_right  is exist
[piper_single_ctrl-1] can_right  is UP
[piper_single_ctrl-1] Error while getting bitrate: [Errno 2] No such file or directory: 'ip'
[piper_single_ctrl-1] Traceback (most recent call last):
[piper_single_ctrl-1]   File "/app/data_collection_ws/install/piper/lib/piper/piper_single_ctrl", line 33, in <module>
[piper_single_ctrl-1]     sys.exit(load_entry_point('piper==0.0.1', 'console_scripts', 'piper_single_ctrl')())
[piper_single_ctrl-1]   File "/app/data_collection_ws/install/piper/lib/python3.10/site-packages/piper/piper_ctrl_single_node.py", line 361, in main
[piper_single_ctrl-1]     piper_single_node = PiperRosNode()
[piper_single_ctrl-1]   File "/app/data_collection_ws/install/piper/lib/python3.10/site-packages/piper/piper_ctrl_single_node.py", line 56, in __init__
[piper_single_ctrl-1]     self.piper = C_PiperInterface(can_name=self.can_port)
[piper_single_ctrl-1]   File "/usr/local/lib/python3.10/dist-packages/piper_sdk/interface/piper_interface.py", line 365, in __init__
[piper_single_ctrl-1]     self.__arm_can=C_STD_CAN(can_name, "socketcan", 1000000, judge_flag, can_auto_init, self.ParseCANFrame)
[piper_single_ctrl-1]   File "/usr/local/lib/python3.10/dist-packages/piper_sdk/hardware_port/can_encapsulation.py", line 59, in __init__
[piper_single_ctrl-1]     self.JudgeCanInfo()
[piper_single_ctrl-1]   File "/usr/local/lib/python3.10/dist-packages/piper_sdk/hardware_port/can_encapsulation.py", line 126, in JudgeCanInfo
[piper_single_ctrl-1]     raise ValueError(f"CAN port {self.channel_name} bitrate is {actual_bitrate} bps, expected {self.expected_bitrate} bps.")
[piper_single_ctrl-1] ValueError: CAN port can_right bitrate is Unknown bps, expected 1000000 bps.
[piper_single_ctrl-1] CAN bus connection was not properly initialized.
[ERROR] [piper_single_ctrl-1]: process has died [pid 5699, exit code 1, cmd '/app/data_collection_ws/install/piper/lib/piper/piper_single_ctrl --ros-args -r __node:=piper_ctrl_single_node --params-file /tmp/launch_params_xz8ce_ww -r joint_ctrl_single:=/joint_states'].

```

Any help would be much appriciated

RosenYin, 2025-02-19:
注意这一行报错：`ValueError: CAN port can_right bitrate is Unknown bps, expected 1000000 bps.`
有可能是激活can模块的时候波特率设定错误；
如果确定没有错误，请在创建interface实例的地方，将`judge_flag`这个参数置为`False`

Akumar201, 2025-02-19:
Hello @RosenYin Thank you for your response, the issue was missing package `iproute2`. After installation of this package using the following command the issue was resolved `sudo apt-get install iproute2`. The issue has been resolved, if the issue persist please feel free re reopen it. 

Thank You

closed 2025-02-19 by Akumar201