agilexrobotics/piper_ros #14: The gripper cannot open entirely

Zhuomin-Zhou, 2025-03-05:
ROS2: humble

Hi,

I noticed the gripper could not open entirely, i.e. half of its limitation, while using the movit2. However, while setting `rviz_ctrl_flag = True` for piper_ctrl_single_node, the gripper operates normally.

Upon reviewing the **piper_ctrl_single_node.py**, I observed that lines 239-240, i.e. `if self.rviz_ctrl_flag: joint_6 = joint_6 * 2`,  apply a conditional doubling of the gripper movement.  After removing this condition (`if self.rviz_ctrl_flag`), Gripper with MoveIt2 functioned correctly.

Regards,
Min

shengyangzhuang, 2025-03-06:
Are you able to use moveit2 to control gazebo (with gripper)? The robot model didn't appear even though I ran moveit2 after launching gazebo.

```
szhuang@alchemy:~/piper_ros_ws$ ros2 launch piper_with_gripper_moveit piper_moveit.launch.py
[INFO] [launch]: All log files can be found below /home/szhuang/.ros/log/2025-03-06-14-33-51-238666-alchemy-67539
[INFO] [launch]: Default logging verbosity is set to INFO
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
[INFO] [move_group-1]: process started with pid [67542]
[INFO] [rviz2-2]: process started with pid [67544]
[move_group-1] [INFO] [1741268031.363100362] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0 seconds
[move_group-1] [INFO] [1741268031.363129068] [moveit_robot_model.robot_model]: Loading robot model 'piper'...
[move_group-1] [INFO] [1741268031.363133018] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[rviz2-2] [INFO] [1741268031.524293786] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-2] [INFO] [1741268031.524332700] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-2] [INFO] [1741268031.533096480] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-2] Warning: class_loader.impl: SEVERE WARNING!!! A namespace collision has occurred with plugin factory for class rviz_default_plugins::displays::InteractiveMarkerDisplay. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one running right now generating this message). Please separate plugins out into their own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.
[rviz2-2]          at line 253 in /opt/ros/humble/include/class_loader/class_loader/class_loader_core.hpp
[move_group-1] [INFO] [1741268031.803500706] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-1] [INFO] [1741268031.803586335] [moveit.ros_planning_interface.moveit_cpp]: Listening to 'joint_states' for joint states
[move_group-1] [INFO] [1741268031.803838475] [moveit_ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
[move_group-1] [INFO] [1741268031.804075822] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[move_group-1] [INFO] [1741268031.804082255] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[move_group-1] [INFO] [1741268031.804138715] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/planning_scene'
[move_group-1] [INFO] [1741268031.804142259] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[move_group-1] [INFO] [1741268031.804201461] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'collision_object'
[move_group-1] [INFO] [1741268031.804258515] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
[move_group-1] [WARN] [1741268031.804446244] [moveit.ros.occupancy_map_monitor.middleware_handle]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
[move_group-1] freeglut (�/�����>): failed to open display ':0'
[move_group-1] Stack trace (most recent call last):
[move_group-1] #9    Object "", at 0xffffffffffffffff, in 
[move_group-1] terminate called after throwing an instance of 'std::system_error'
[move_group-1]   what():  Owner died
[move_group-1] Stack trace (most recent call last) in thread 67639:
[move_group-1] #19   Object "", at 0xffffffffffffffff, in 
[move_group-1] #8    Object "/opt/ros/humble/lib/moveit_ros_move_group/move_group", at 0x5b5df470a724, in _start
[move_group-1] #18   Source "../sysdeps/unix/sysv/linux/x86_64/clone3.S", line 81, in __clone3 [0x71fc7bd2684f]
[move_group-1] #17   Source "./nptl/pthread_create.c", line 442, in start_thread [0x71fc7bc94ac2]
[move_group-1] #16   Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x71fc7c0dc252, in 
[move_group-1] #15   Object "/opt/ros/humble/lib/librclcpp.so", at 0x71fc7c5309e0, in rclcpp::executors::SingleThreadedExecutor::spin()
[move_group-1] #14   Object "/opt/ros/humble/lib/librclcpp.so", at 0x71fc7c529432, in rclcpp::Executor::get_next_executable(rclcpp::AnyExecutable&, std::chrono::duration<long, std::ratio<1l, 1000000000l> >)
[move_group-1] #13   Object "/opt/ros/humble/lib/librclcpp.so", at 0x71fc7c52672b, in rclcpp::Executor::wait_for_work(std::chrono::duration<long, std::ratio<1l, 1000000000l> >)
[move_group-1] #12   Object "/opt/ros/humble/lib/librcl.so", at 0x71fc7b9dd847, in rcl_wait
[move_group-1] #11   Object "/opt/ros/humble/lib/librmw_fastrtps_cpp.so", at 0x71fc789e2886, in rmw_wait
[move_group-1] #10   Object "/opt/ros/humble/lib/librmw_fastrtps_shared_cpp.so", at 0x71fc7898a5eb, in rmw_fastrtps_shared_cpp::__rmw_wait(char const*, rmw_subscriptions_s*, rmw_guard_conditions_s*, rmw_services_s*, rmw_clients_s*, rmw_events_s*, rmw_wait_set_s*, rmw_time_s const*)
[move_group-1] #9    Object "/opt/ros/humble/lib/libfastrtps.so.2.6.9", at 0x71fc783e7f9c, in eprosima::fastdds::dds::detail::WaitSetImpl::attach_condition(eprosima::fastdds::dds::Condition const&)
[... cut ...]
[ERROR] [move_group-1]: process has died [pid 67542, exit code -6, cmd '/opt/ros/humble/lib/moveit_ros_move_group/move_group --ros-args --params-file /tmp/launch_params_nzemx5p5 --params-file /tmp/launch_params_zqz3le9v --params-file /tmp/launch_params_pi4pwbrs'].
[rviz2-2] [ERROR] [1741268034.588437325] [moveit_ros_visualization.motion_planning_frame]: Action server: /recognize_objects not available
[rviz2-2] [INFO] [1741268034.602547135] [moveit_ros_visualization.motion_planning_frame]: MoveGroup namespace changed: / -> . Reloading params.

```

HoranTian, 2025-03-07:
> 您能够使用 moveit2 来控制凉亭（带夹持器）吗？即使我在启动 gazebo 后运行 moveit2，机器人模型也没有出现。
> 
> ```
> szhuang@alchemy:~/piper_ros_ws$ ros2 launch piper_with_gripper_moveit piper_moveit.launch.py
> [INFO] [launch]: All log files can be found below /home/szhuang/.ros/log/2025-03-06-14-33-51-238666-alchemy-67539
> [INFO] [launch]: Default logging verbosity is set to INFO
> Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> [INFO] [move_group-1]: process started with pid [67542]
> [INFO] [rviz2-2]: process started with pid [67544]
> [move_group-1] [INFO] [1741268031.363100362] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0 seconds
> [move_group-1] [INFO] [1741268031.363129068] [moveit_robot_model.robot_model]: Loading robot model 'piper'...
> [move_group-1] [INFO] [1741268031.363133018] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
> [rviz2-2] [INFO] [1741268031.524293786] [rviz2]: Stereo is NOT SUPPORTED
> [rviz2-2] [INFO] [1741268031.524332700] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
> [rviz2-2] [INFO] [1741268031.533096480] [rviz2]: Stereo is NOT SUPPORTED
> [rviz2-2] Warning: class_loader.impl: SEVERE WARNING!!! A namespace collision has occurred with plugin factory for class rviz_default_plugins::displays::InteractiveMarkerDisplay. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one running right now generating this message). Please separate plugins out into their own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.
> [rviz2-2]          at line 253 in /opt/ros/humble/include/class_loader/class_loader/class_loader_core.hpp
> [move_group-1] [INFO] [1741268031.803500706] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
> [move_group-1] [INFO] [1741268031.803586335] [moveit.ros_planning_interface.moveit_cpp]: Listening to 'joint_states' for joint states
> [move_group-1] [INFO] [1741268031.803838475] [moveit_ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
> [move_group-1] [INFO] [1741268031.804075822] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
> [move_group-1] [INFO] [1741268031.804082255] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
> [move_group-1] [INFO] [1741268031.804138715] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/planning_scene'
> [move_group-1] [INFO] [1741268031.804142259] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
> [move_group-1] [INFO] [1741268031.804201461] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'collision_object'
> [move_group-1] [INFO] [1741268031.804258515] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
> [move_group-1] [WARN] [1741268031.804446244] [moveit.ros.occupancy_map_monitor.middleware_handle]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
> [move_group-1] freeglut (�/�����>): failed to open display ':0'
> [move_group-1] Stack trace (most recent call last):
> [move_group-1] #9    Object "", at 0xffffffffffffffff, in 
> [move_group-1] terminate called after throwing an instance of 'std::system_error'
> [move_group-1]   what():  Owner died
> [move_group-1] Stack trace (most recent call last) in thread 67639:
> [move_group-1] #19   Object "", at 0xffffffffffffffff, in 
> [move_group-1] #8    Object "/opt/ros/humble/lib/moveit_ros_move_group/move_group", at 0x5b5df470a724, in _start
> [move_group-1] #18   Source "../sysdeps/unix/sysv/linux/x86_64/clone3.S", line 81, in __clone3 [0x71fc7bd2684f]
> [move_group-1] #17   Source "./nptl/pthread_create.c", line 442, in start_thread [0x71fc7bc94ac2]
> [move_group-1] #16   Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x71fc7c0dc252, in 
> [move_group-1] #15   Object "/opt/ros/humble/lib/librclcpp.so", at 0x71fc7c5309e0, in rclcpp::executors::SingleThreadedExecutor::spin()
> [move_group-1] #14   Object "/opt/ros/humble/lib/librclcpp.so", at 0x71fc7c529432, in rclcpp::Executor::get_next_executable(rclcpp::AnyExecutable&, std::chrono::duration<long, std::ratio<1l, 1000000000l> >)
> [move_group-1] #13   Object "/opt/ros/humble/lib/librclcpp.so", at 0x71fc7c52672b, in rclcpp::Executor::wait_for_work(std::chrono::duration<long, std::ratio<1l, 1000000000l> >)
> [move_group-1] #12   Object "/opt/ros/humble/lib/librcl.so", at 0x71fc7b9dd847, in rcl_wait
> [move_group-1] #11   Object "/opt/ros/humble/lib/librmw_fastrtps_cpp.so", at 0x71fc789e2886, in rmw_wait
> [move_group-1] #10   Object "/opt/ros/humble/lib/librmw_fastrtps_shared_cpp.so", at 0x71fc7898a5eb, in rmw_fastrtps_shared_cpp::__rmw_wait(char const*, rmw_subscriptions_s*, rmw_guard_conditions_s*, rmw_services_s*, rmw_clients_s*, rmw_events_s*, rmw_wait_set_s*, rmw_time_s const*)
> [move_group-1] #9    Object "/opt/ros/humble/lib/libfastrtps.so.2.6.9", at 0x71fc783e7f9c, in eprosima::fastdds::dds::detail::WaitSetImpl::attach_condition(eprosima::fastdds::dds::Condition const&)
[... cut ...]
> [ERROR] [move_group-1]: process has died [pid 67542, exit code -6, cmd '/opt/ros/humble/lib/moveit_ros_move_group/move_group --ros-args --params-file /tmp/launch_params_nzemx5p5 --params-file /tmp/launch_params_zqz3le9v --params-file /tmp/launch_params_pi4pwbrs'].
> [rviz2-2] [ERROR] [1741268034.588437325] [moveit_ros_visualization.motion_planning_frame]: Action server: /recognize_objects not available
> [rviz2-2] [INFO] [1741268034.602547135] [moveit_ros_visualization.motion_planning_frame]: MoveGroup namespace changed: / -> . Reloading params.
> ```

Try to run the demo.launch.py

HoranTian, 2025-03-07:
> ROS2：谦逊
> 
> 你好
> 
> 我注意到在使用 movit2 时，抓手无法完全打开，即其限制的一半。但是，在设置为 piper_ctrl_single_node 时，机械手会正常运行。`rviz_ctrl_flag = True`
> 
> 在回顾**piper_ctrl_single_node.py**时，我观察到第 239-240 行，即 ，应用了机械手运动的有条件加倍。删除此条件 （） 后，带有 movit2 的夹持器正常工作。`if self.rviz_ctrl_flag: joint_6 = joint_6 * 2``if self.rviz_ctrl_flag`
> 
> 此致， Min

We have updated the parameters of start_single_piper.launch.py, using gripper_val_multiple to control the gripper multiplier. The reason is mentioned in the readme, as moveit requires the gripper to be set to double when controlling the real robotic arm.

[image]

[image]

shengyangzhuang, 2025-03-07:
> Try to run the demo.launch.py

Thanks for your reply. 

So I mean running the gazebo simulation and use moveit2 to control the robot in the simulation.

I think demo.launch.py is used for real robot.

Will you have similar issue running the simulation? You may try the following:

'''
ros2 launch piper_gazebo piper_gazebo.launch.py
'''
'''
ros2 launch piper_with_gripper_moveit piper_moveit.launch.py 
'''

HoranTian, 2025-03-07:
> > 尝试运行 demo.launch.py
> 
> 感谢您的回复。
> 
> 所以我的意思是运行 gazebo 模拟并使用 moveit2 来控制模拟中的机器人。
> 
> 我认为 demo.launch.py 用于真正的机器人。
> 
> 您在运行模拟时会遇到类似的问题吗？您可以尝试以下方法：
> 
> ```
> ros2 launch piper_gazebo piper_gazebo.launch.py
> ```
> 
> ```
> ros2 launch piper_with_gripper_moveit piper_moveit.launch.py 
> ```

Yes, demo.launch.py can control the real robotic arm, but it uses the same URDF as piper_moveit.launch.py. 
You can try running the demo to check if the model loading failure is due to the robot_description not being loaded successfully or if there are issues with the URDF and SRDF writing. 
This is because the demo can publish the robot_description, while piper_moveit.launch relies on Gazebo to publish the robot_description. 
Below are the results of my execution of piper_moveit.launch.

[image]

[image]

The left side of the second image is the gazebo, and the right side is MoveIt.

HoranTian, 2025-03-12:
> > 是的，demo.launch.py 可以控制真正的机械臂，但它使用与 piper_moveit.launch.py相同的 URDF。您可以尝试运行演示，以检查模型加载失败是由于robot_description未成功加载造成的，还是由于 URDF 和 SRDF 写入存在问题。这是因为 demo 可以发布 robot_description，而 piper_moveit.launch 依赖于 Gazebo 来发布robot_description。以下是我执行 piper_moveit.launch 的结果。
> 
> 感谢您的想法。今天，我运行了真正的机器人，它与 moveit2 配合得很好，这意味着可以正确读取 SRDF。但是，当我使用 moveit2 运行凉亭时，它再次显示相同的错误。它仍然指示无法打开显示设备，找不到参数robot_description_semantic，无法解析 SRDF XML 文件，作服务器 /recognize_objects 不可用。
> 
> 奇怪的是，我可以用 moveit2 正确运行真正的机器人，用 moveit2 运行凉亭机器人（没有夹爪）。在将凉亭机器人（带夹爪）与 moveit2 一起使用时，似乎没有理由出现上述错误，这让我感到困惑......更重要的是，它对您来说效果很好，而且我们使用的是相同的软件包。
> 
> ```
> szhuang@alchemy:~/piper_ros_ws$ ros2 launch piper_with_gripper_moveit piper_moveit.launch.py
> [INFO] [launch]: All log files can be found below /home/szhuang/.ros/log/2025-03-10-16-33-12-939706-alchemy-20646
> [INFO] [launch]: Default logging verbosity is set to INFO
> Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
> [INFO] [move_group-1]: process started with pid [20647]
> [INFO] [rviz2-2]: process started with pid [20649]
> [move_group-1] [INFO] [1741620793.064041168] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0 seconds
> [move_group-1] [INFO] [1741620793.064072557] [moveit_robot_model.robot_model]: Loading robot model 'piper'...
> [move_group-1] [INFO] [1741620793.064076113] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
> [rviz2-2] [INFO] [1741620793.223140596] [rviz2]: Stereo is NOT SUPPORTED
> [rviz2-2] [INFO] [1741620793.223182843] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
> [rviz2-2] [INFO] [1741620793.232560356] [rviz2]: Stereo is NOT SUPPORTED
> [rviz2-2] Warning: class_loader.impl: SEVERE WARNING!!! A namespace collision has occurred with plugin factory for class rviz_default_plugins::displays::InteractiveMarkerDisplay. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one running right now generating this message). Please separate plugins out into their own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.
> [rviz2-2]          at line 253 in /opt/ros/humble/include/class_loader/class_loader/class_loader_core.hpp
> [move_group-1] [INFO] [1741620793.502477430] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
> [move_group-1] [INFO] [1741620793.502546737] [moveit.ros_planning_interface.moveit_cpp]: Listening to 'joint_states' for joint states
> [move_group-1] [INFO] [1741620793.502812661] [moveit_ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
> [move_group-1] [INFO] [1741620793.503181356] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
> [move_group-1] [INFO] [1741620793.503190240] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
> [move_group-1] [INFO] [1741620793.503270606] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/planning_scene'
> [move_group-1] [INFO] [1741620793.503274994] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
> [move_group-1] [INFO] [1741620793.503343753] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'collision_object'
> [move_group-1] [INFO] [1741620793.503399641] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
> [move_group-1] [WARN] [1741620793.503559953] [moveit.ros.occupancy_map_monitor.middleware_handle]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
> [move_group-1] freeglut (�/�����>): failed to open display ':0'
> [move_group-1] Stack trace (most recent call last):
> [move_group-1] #9    Object "", at 0xffffffffffffffff, in 
> [ERROR] [move_group-1]: process has died [pid 20647, exit code -11, cmd '/opt/ros/humble/lib/moveit_ros_move_group/move_group --ros-args --params-file /tmp/launch_params_lp8donoc --params-file /tmp/launch_params_kppwu58x --params-file /tmp/launch_params_p8fxtsrq'].
> [rviz2-2] [ERROR] [1741620796.291873747] [moveit_ros_visualization.motion_planning_frame]: Action server: /recognize_objects not available
> [rviz2-2] [INFO] [1741620796.309028296] [moveit_ros_visualization.motion_planning_frame]: MoveGroup namespace changed: / -> . Reloading params.
> [rviz2-2] [ERROR] [1741620806.378928675] [rviz]: Could not find parameter robot_description_semantic and did not receive robot_description_semantic via std_msgs::msg::String subscription within 10.000000 seconds.
> [rviz2-2] Error:   Could not parse the SRDF XML File. Error=XML_ERROR_EMPTY_DOCUMENT ErrorID=13 (0xd) Line number=0
> [rviz2-2]          at line 732 in ./src/model.cpp
> [rviz2-2] [ERROR] [1741620806.382310323] [moveit_rdf_loader.rdf_loader]: Unable to parse SRDF
> [rviz2-2] [ERROR] [1741620806.394753011] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Robot model not loaded
> ```

We have retested in Docker and configured a new environment, which is functioning properly. It is recommended to check the system environment or reconfigure using the system environment.

closed 2025-03-27 by HoranTian