agilexrobotics/piper_ros #19: 关于频率问题

YanJiaHuan, 2025-03-24:
在piper_ctr_single_node.py 里的 [publish_thread](https://github.com/agilexrobotics/piper_ros/blob/humble/src/piper/piper/piper_ctrl_single_node.py#L81)函数，应该是定频200hz发布/end_pose, /joint_state_single这些信息。
但是通过 `ros2 topic hz`这个指令看两个节点，都是接近400hz。这是什么原因？

**ros2 topic hz /end_pose**
average rate: 399.782
        min: 0.000s max: 0.005s std dev: 0.00079s window: 401
average rate: 400.038
        min: 0.000s max: 0.006s std dev: 0.00086s window: 802
average rate: 399.818
        min: 0.000s max: 0.006s std dev: 0.00088s window: 1202
**ros2 topic hz /joint_states_single**
average rate: 200.080
        min: 0.004s max: 0.006s std dev: 0.00027s window: 202
average rate: 293.589
        min: 0.000s max: 0.006s std dev: 0.00147s window: 590
average rate: 328.946
        min: 0.000s max: 0.006s std dev: 0.00139s window: 990
average rate: 346.664
        min: 0.000s max: 0.006s std dev: 0.00133s window: 1390

我启动piper单个机械臂的命令：
`ros2 launch piper start_single_piper.launch.py can_port:=can_piper auto_enable:=true gripper_exist:=false gripper_val_mutiple:=2`

HoranTian, 2025-03-27:
[image]
测试的是保持在200hz的，请更新sdk并重新获取ros尝试

YanJiaHuan, 2025-03-27:
断电重连现在没问题了，也有可能是我自己启动了两次

closed 2025-03-27 by YanJiaHuan