agilexrobotics/piper_ros #20: piper ros on raspberry pi 5

Crypto69, 2025-04-21:
Can you run Piper on a raspberry pi 5?

Seems the newest piper ros release is based on humble which only supports ubuntu 22.04 which does not run on a raspberry pi 5.

Any suggestions what to do to run piper_ros  on a raspberry pi 5?

RosenYin, 2025-04-22:
piper ros有两个版本，一个是基于noetic一个是基于humble，这两个基本都是同步更新的，humble无法使用可以尝试使用noetic版本

Crypto69, 2025-04-23:
Are there any plans to support Jazzy Jalisco on piper_ros?  It has native support for Ubuntu 24.x

noetic only supported on Ubuntu 20.04. 

This means the only way to support noetic or humble is to use a docker image. Is there another way I am missing?

是否有计划在 piper ros 上支持 Jazzy Jalisco？它对 Ubuntu 24.x 提供原生支持。

noetic 仅支持 Ubuntu 20.04。

这意味着支持 noetic 或 humble 的唯一方法是使用 Docker 镜像。我是否遗漏了其他方式？